thonny/thonny #1945: Importing module is hit or miss.

icancwhatusay, 2021-08-23:
Writing the following in Notepadd++ works fine 

```
import pyautogui as py

py.typewrite("bunch of words")
```

Writing the exact same code in Thonny does not. I get :

[image]

Even skipping the 'as py' part I get the same error.

In case the screen shot didn't stick, here's the error message:

```
>>> %Run 11.py
Traceback (most recent call last):
  File "C:\Users\************\11.py", line 1, in <module>
    import pyautogui
ModuleNotFoundError: No module named 'pyautogui'
>>> 
```

aivarannamaa, 2021-08-23:
If you can import a module when running from Notepad++ and can't when running from Thonny, then you are using different Python interpreter in either case. 

I assume you installed Thonny via binary Thonny+Python bundle (eg. https://github.com/thonny/thonny/releases/download/v3.3.13/thonny-3.3.13.exe). In this case Thonny is running your programs in its own Python interpreter by default. You can either install required packages for this interpreter, or switch to an external interpreter (with the menu button in the lower-right corner of the main window)

aivarannamaa, 2021-11-06:
Please reopen if you still have this problem!

closed 2021-11-06 by aivarannamaa