thonny/thonny #1946: thonny is not working

Shafikulachalil, 2021-08-24:
Previously i used thonny with my ubuntu 18.04 ,now I can't open , if I tried to open thonny this is the output,how to solve this??
i tried reinstall thonny ,but no use
```
NFO    thonny: Thonny version: 3.3.14
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 183, in launch
    _delegate_to_existing_instance(sys.argv[1:])
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 278, in _delegate_to_existing_instance
    sock, secret = _create_client_socket()
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 321, in _create_client_socket
    client_socket.connect(get_ipc_file_path())
ConnectionRefusedError: [Errno 111] Connection refused
No protocol specified
No protocol specified
ERROR:root:Internal launch or mainloop error
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 196, in launch
    bench = workbench.Workbench()
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/workbench.py", line 131, in __init__
    tk.Tk.__init__(self, className="Thonny")
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 196, in launch
    bench = workbench.Workbench()
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/workbench.py", line 131, in __init__
    tk.Tk.__init__(self, className="Thonny")
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/thonny", line 33, in <module>
    sys.exit(load_entry_point('thonny==3.3.14', 'gui_scripts', 'thonny')())
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/__init__.py", line 216, in launch
    dlg = ui_utils.LongTextDialog("Internal error", traceback.format_exc())
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/ui_utils.py", line 1740, in __init__
    super().__init__(master=parent)
  File "/usr/local/lib/python3.6/dist-packages/thonny-3.3.14-py3.6.egg/thonny/ui_utils.py", line 41, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2342, in __init__
    BaseWidget.__init__(self, master, 'toplevel', cnf, {}, extra)
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2292, in __init__
    BaseWidget._setup(self, master, cnf)
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2259, in _setup
    _default_root = Tk()
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"

```

Shafikulachalil, 2021-09-01:
Thanks for the replay,  what is XDG_RUNTIME_DIR??
how to access ipc.sock??

Shafikulachalil, 2021-09-01:
showing " No such file or directory"

aivarannamaa, 2021-11-04:
Sorry for late reply!

Did you manage to get Thonny working?

It looks like you have pip-installed Thonny. You could try Python+Thonny bundle instead -- for this execute following on your shell prompt:

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```

closed 2021-11-05 by aivarannamaa