thonny/thonny #1947: Where to save to? popup off screen not visible

guser210, 2021-08-24:
Where to save to? popup does not appear when you have two displays with different resolutions in windows 10. I have my main display resolution at 3840 x 2400 and a 2nd display at 1080, when the Thonny is in the 2nd display either maximized or restore all popups appear off both screens, forcing me to use Alt + space and the arrow keys to move it back into view. this does not happen when Thonny is in the main screen.

Thonny 3.3.13
System: Windows 10 64bit
Python 3.7.9 32bit
Tk 8.6.9

aivarannamaa, 2022-03-24:
Please try whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works for you

guser210, 2022-05-16:
Thank you, it fixed the issue.

closed 2022-05-16 by guser210