thonny/thonny #1953: Thonny UI does not show Theme color

islaloha, 2021-09-01:
Thonny UI only shows black and white and doesn't show the theme color. It is always black and white. I am using the latest version in Windows 10. Any idea? Thanks.

aivarannamaa, 2021-09-16:
Sorry for the late reply! 

Adapting to system theme is not implemented in Thonny.

closed 2021-11-06 by aivarannamaa

icarus7776, 2024-08-27:
Hi, I know this is an old post and not sure if this will be seen but I'm having the same issue - I have Thonny installed on a pc and a Surface Pro. The PC shows all the theme colors but the Surface Pro shows black and white. In the Surface I can even change the theme and see the preview update but the editor stays black and white. In the Editor settings, I've ensured all the highlight settings are checked. I've even uninstalled/reinstalled Thonny. It just stays black and white. I can't work out why!