thonny/thonny #1956: FYI: starting thonny leads to ModuleNotFoundError: No module named 'tkinter'. Solution: install python3-tk

sanderjo, 2021-09-02:
As a FYI:

**Short**

starting thonny leads to `ModuleNotFoundError: No module named 'tkinter'`
Solution: `sudo apt-get install python3-tk`

**Long**

Platform: Ubuntu on ARM64 / Aarch64 ...headless

After a `python3 -m pip -v install thonny`

and
```
...
Successfully installed Send2Trash-1.8.0 astroid-2.7.3 asttokens-2.0.5 docutils-0.17.1 isort-5.9.3 jedi-0.18.0 lazy-object-proxy-1.6.0 mccabe-0.6.1 mypy-0.910 mypy-extensions-0.4.3 parso-0.8.2 platformdirs-2.3.0 pylint-2.10.2 pyserial-3.5 setuptools-57.4.0 six-1.16.0 thonny-3.3.14 toml-0.10.2 typed-ast-1.4.3 typing-extensions-3.10.0.2 wrapt-1.12.1
Cleaning up...
```
So far so good, but starting thonny leads to `ModuleNotFoundError: No module named 'tkinter'`

Solution: `sudo apt-get install python3-tk`

```
sander@nanopineo2:~$ thonny
INFO    thonny: Thonny version: 3.3.14
Traceback (most recent call last):
  File "/home/sander/.local/bin/thonny", line 11, in <module>
    sys.exit(launch())
  File "/home/sander/.local/lib/python3.6/site-packages/thonny/__init__.py", line 178, in launch
    if not _check_welcome():
  File "/home/sander/.local/lib/python3.6/site-packages/thonny/__init__.py", line 146, in _check_welcome
    from thonny.config import ConfigurationManager
  File "/home/sander/.local/lib/python3.6/site-packages/thonny/config.py", line 9, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'
sander@nanopineo2:~$
```

aivarannamaa, 2021-09-02:
I'm not sure I understood properly -- did you manage to get tkinter/thonny working or not?

sanderjo, 2021-09-02:
> I'm not sure I understood properly -- did you manage to get tkinter/thonny working or not?

It's a FYI to solve that dependency. Afterwards it's obvious, but I didn't see it in the documentation.

aivarannamaa, 2021-09-28:
Thanks for the nudge! I now updated Linux installation instructions: https://github.com/thonny/thonny/wiki/Linux#installing-with-pip

closed 2021-09-28 by aivarannamaa