thonny/thonny #1957: Any way to set up a classroom installation with some class-wide packages, but students can still install more?

grodstein, 2021-09-02:
We're trying to set up Thonny for a classroom such that:

- One installation done by our school's IT group and used by all students in the class
- In addition to the normal packages that come with Thonny, the installation also sets up numpy, matplotlib and a few more. This should be done just once by the IT group
- Students can still install a few more packages on their own if they want to learn about them.

The issue is that the main Thonny installation and all of the common packages must be in a protected directory, to avoid having students mistakenly break the installation. But then we don't see a good way to have students able to install anything.

Are we missing something about the installation process? We've looked at https://github.com/thonny/thonny/wiki/DeploymentOptions but don't see anything that looks like it will solve our problem. I.e., we don't see any way to have numpy, matplotlib, etc. in the Thonny program folder and also have a Thonny data folder available for each student.

Thanks,
/Joel

aivarannamaa, 2021-09-03:
After installing Thonny, you could install required packages into Thonny's main site-packages directory. For example, if Thonny is installed to `C:\Program Files\Thonny`, then you could execute

```
> "C:\Program Files\Thonny\python.exe" -m pip install numpy matplotlib
```

The users could install extra packages via Thonny's package manager, which uses `pip install --user ...` under hood.

grodstein, 2021-09-03:
Makes sense. Thanks!
/Joel

closed 2021-09-03 by grodstein