thonny/thonny #1958: Not working with Adafruit CircuitPython 7.0.0-rc.1 on RP2040

LovelyA72, 2021-09-03:
When I connect RP2040 board with CircuitPython 7.0.0-RC.1loaded with Thonny, I only got an empty console and an empty file explorer. ctrl+c does nothing in the console. The board will have its white LED on to confirm that it's connected to the console but I can't send out any commands from the console.

The console also occasionally prints
```
WARNING:root:Unexpected echo. Expected b'help\r\n', got b''
```

However, This might be an issue with Adafruit's CircuitPython firmware

aivarannamaa, 2021-09-03:
Thank you for the report! I'll try to find time for reproducing this in the weekend

aivarannamaa, 2021-09-06:
I think I found the reason -- CP7 has removed a marker string which is used by Thonny as confirmation about entering raw REPL (https://github.com/adafruit/circuitpython/issues/5307). Hopefully they put it back before final release.

carlskii, 2021-09-07:
+1 exactly the same problem for me.

aivarannamaa, 2021-11-06:
closing, as the final CP 7.0 fixed it

closed 2021-11-06 by aivarannamaa