thonny/thonny #1959: Display of program

Hermann12, 2021-09-04:
I realized that Thonny have a problem with display Checkbutton in tkinter:
[image]
If I execute the same souce with Spyder I will get this nicer view, why?
[image]

aivarannamaa, 2021-09-04:
Thank you for the report!

In Windows, Thonny calls `SetProcessDpiAwareness(1)` when initializing the process for user code. Unfortunately Tk prior to 8.6.11 has problems with showing certain widgets with high dpi.

The best I could do is make `SetProcessDpiAwareness(1)` optional, but as Thonny 4.0 will come with Tk 8.6.11, I'd rather not introduce another option.

You could try adding following in the beginning of your code:

```python
import ctypes
ctypes.OleDLL("shcore").SetProcessDpiAwareness(0)
```

If you're interested, then here is an experimental bundle of Thonny 3.3.7 containing Tk 8.6.11: https://github.com/thonny/thonny/releases/download/v3.3.7/thonny-3.3.7-alt.exe

Hermann12, 2021-09-04:
With this lines I get this Error message:
ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Traceback (most recent call last):
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1238, in wrapper
    result = method(self, *args, **kwargs)
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1225, in wrapper
    return method(self, *args, **kwargs)
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1295, in _execute_prepared_user_code
    exec(statements, global_vars)
  File "D:\Daten\Programmieren\Fritz_04\Mini.py", line 86, in <module>
    main()
  File "D:\Daten\Programmieren\Fritz_04\Mini.py", line 71, in main
    ctypes.OleDLL("shcore").SetProcessDpiAwareness(0)
  File "_ctypes/callproc.c", line 997, in GetResult
PermissionError: [WinError -2147024891] Zugriff verweigert

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1283, in execute_source
    return self._execute_prepared_user_code(statements, global_vars)
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1243, in wrapper
    return {"user_exception": self._backend._prepare_user_exception()}
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1076, in _prepare_user_exception
    "stack": self._export_stack(last_frame),
  File "c:\python39\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 968, in _export_stack
    module_name = system_frame.f_globals["__name__"]
KeyError: '__name__'

aivarannamaa, 2021-11-04:
Sorry for the late reply!

I don't have any ideas what to try next, but with the 4.0 approaching, I'll close this issue.

closed 2021-11-04 by aivarannamaa

Hermann12, 2022-03-26:
Hope it will be work better, because also in Thonny the check box is not readable very good. 
[image]

aivarannamaa, 2022-04-10:
Please try https://github.com/thonny/thonny/releases/tag/v4.0.0b2, which has newer Tk

Hermann12, 2022-04-11:
I use thonny with pip install ..... How can I update into python environment to V4.0.0b2 ? THANKS!

aivarannamaa, 2022-04-11:
Latest pre-releases can be installed with `--pre` switch: `pip install --pre thonny`

Hermann12, 2022-04-15:
I tested this issue with V4 beta, but no difference. Anyway thanks! I use python 3.9.8.
[image]

aivarannamaa, 2022-04-23:
>  I use python 3.9.8.

If you can then try upgrading to 3.9.9 or later. According to Python changelog, they included newer Tk in 3.9.9, which should fix this issue. 

In addition to this, I decided to turn off automatic SetProcessDpiAwareness for user programs starting with Thonny 4.0.0b3. See #2159 for more.

Hermann12, 2022-05-02:
Python 3.9.11 and Thonny V4 preview doesn't show this weak point in tkinter anymore. Thank you for your great support!