thonny/thonny #1962: [tutorial] add windows explorer context menu "edit with thonny" for .py files

atesin, 2021-09-05:
this feature could be useful if you code a lot of python files... i was searching for a tool for editing explorer context menus easily but none i tested worked me fine, until i realized doing it directly with regedit is easy and quick enough... here goes

disclaimer: modifying the windows registry could be dangerous for your computer, do it carefully or if unsure better don't

- open regedit, in left pane find HKEY_CLASSES_ROOT key
- search for '.py' subkey
- open and take note the default name at right pane, mine was "Python.File"
- copy the text below and paste in your favorite editor (notepad, madedit, geany, vim, etc)
```
Windows Registry Editor Version 5.00
[HKEY_CLASSES_ROOT\Python.File\Shell\EditWithThonny]
"MUIVerb"="Edit with Thonny"
[HKEY_CLASSES_ROOT\Python.File\Shell\EditWithThonny\command]
@="\"C:\\Users\\YourLocalProfileName\\AppData\\Local\\Programs\\Python\\Python39\\Scripts\\thonny.exe\" -m thonny %*"
```
- modify "Python.File" and "YourLocalProfileName" to match yours
- note the '...thonny.exe' path should match your phython/thonny installation folder, could be 'C:\\Program files\\...' or anywhere, check it and edit before saving
- save the file somewhere with .reg extension, like 'somename.reg' and double click
- test right-clicking some .py file... if worked you can now safe delete de .reg file

aivarannamaa, 2021-09-16:
Thank you!

I need to review Windows installer. I thought that it already does something like this.

atesin, 2021-09-17:
hi...  you could also add a cascade style subbenu, what is actually what i did

[image]

as ain image speaks more than thousand words, this is a shot of my regedit

[image]

- there was already a 'edit39' key for 'edit with idle' cascade submenu entry, if your new subkey has a 'smaller' name it will appear before previous or with 'greater' name will appear after.... mine is 'edit29' so it appears before (see picture above)
- you can also even create another new additional cascade menu 'edithwithother' or whatever, following this same structure, which must have the string 'MUIVerb'='&Edit with'.... it seems the ampersand '&'tells explorer is a cascaded menu

aivarannamaa, 2022-05-21:
@atesin , how did you install Thonny? As admin under Program Files or as regular user under home?

According to my experiments, latest Thonny installers still create "Edit with Thonny" context menu (just like early ones did).

atesin, 2022-05-21:
i installed thonny in my laptop (windows), i am in my pc desktop right now

as i remember, i installed thonny with "pip install thonnyapp" from my normal cmd console window

aivarannamaa, 2022-05-21:
Oh, in case you pip-installed it, then I can close this issue.

closed 2022-05-21 by aivarannamaa