thonny/thonny #1963: [suggestion] options to customize the <tab> key char and size behavior

atesin, 2021-09-05:
hi, thanks for thonny, it rocks

i always felt <tab> character too wide (8 chars!?), vertical alignment sometimes gets too 'distanced'... especially when you have a lot of nested functions and blocks, the code tends to shift to the right, making it hard to navigate sometimes

i would like a couple options to customize the <tab> key behavior, like choose insert <tabs> or <spaces> and the size, like some other editors (but not so good as thonny) have... for example i found my <tab> key looks good inserting 2 spaces... it would be nice :)

aivarannamaa, 2021-09-16:
Thank you for the report!

Which Thonny version are you using and on which platform? At least Thonny 3.3.13 on Ubuntu displays tabs as 4 spaces in the editor.

atesin, 2021-09-16:
hi, yes that is i am talking about.... it displays just 4 chars **fixed**, i would like an option to customize tab width.... this is how code looks with 2 (manual) spaces, you see it looks good and easy to the eyes

[image]
(*i wrote this function, i am learning python and its tricks*)

my specs:

- windows 8.1 pro x64 (*)
- python 3.9.7
- thonny 3.3.14 installed with 'pip install thonnyapp' (and with context menu **)

(*) everybody are crazy with win10 but i dont like it, it is too heavy and intrusive, i would still use win7 but there are features that dont work anymore so i felt forced to upgrade just a few versions enough

(**) 'edit with thonny' explorer context menu: #1962

aivarannamaa, 2021-09-16:
I'll consider adding a hidden setting for tab size.

> i would still use win7 but there are features that dont work anymore so i felt forced to upgrade just a few versions enough

Good idea, as starting with version 3.9, Python doesn't support Win7 anymore.

atesin, 2021-09-17:
"won't support" == "won't run" ???   (sorry for the off topic, couldnt resist xD )

thanks  for consider it ...  i have more ideas i will put here eventually ....  for example, do you know about 'scintilla' text editor engine used by notepad++ and others???  ....  it is **fullteatured**, maybe would be good if you read a little about it

https://en.wikipedia.org/wiki/Scintilla_(software)

aivarannamaa, 2021-09-17:
> "won't support" == "won't run" ??? (sorry for the off topic, couldnt resist xD )

Won't run. The installer refuses to continue, and the Python executable from the zipped distributions exits with errors.

I know Scintilla, but it's not easy to integrate it into Tk(inter) based UI. At least in a cross-platform and easily distributable way.

aivarannamaa, 2024-05-30:
indent-width, tab-width and indent with tab options have been implemented in  #1460

closed 2024-05-30 by aivarannamaa