thonny/thonny #1964: Cannot connect to FireBeetle

jnko, 2021-09-06:
Thonny 3.3.14, set to MicroPython (ESP32), Port: Auto
Python 3.9.6 (local), MicroPython 1.16 (device)
Device: DFRobot FireBeetle ESP32 V4.0 (DFR0478)

For some reason I am unable to connect or run current script on the device.
The Thonny-Shell is empty and typing anything leads to
'''
print("...")
WARNING:root:Unexpected echo. Expected b'print("...")\r\n', got b''
'''

However placing a small script into boot.py produces output in Thonny.
'''
#boot.py
print("boot.py is alive")
'''

Now resetting/starting the ESP32 lets Thonny show the output but missing the repl '>>>'
'''
boot.py is alive
'''

Notable to say that other boards are working without problems, for example HUZZAH32 or PyCom.
Anyway I'm able to connect to the FireBeetle board within a terminal (ttyUSB0) without any problems.
Using other (oder) MicryPython versions does not help.

aivarannamaa, 2021-09-06:
Thank you for the report!

Are you using a specific Firebeetle build of MicroPython or the generic ESP32 one?

jnko, 2021-09-06:
It is a generic ESP32 MicroPython build downloaded from micropython.org.

https://micropython.org/resources/firmware/esp32-20210623-v1.16.bin

(Sorry short in words, using my mobile right now)

aivarannamaa, 2021-09-21:
I received my FireBeetle and was able to reproduce the problem in Ubuntu. It may be relevant that I could _not_ reproduce the problem neither in Windows nor macOS. This means the USB driver might be involved in this issue.

aivarannamaa, 2021-09-21:
I found a work-around: read https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration and set your `submit_mode = paste`.

Notes for myself: It looks like Thonny is able to execute some management commands on FireBeetle, but when it reaches the command which is meant to output a short string (the value of `sys.path`), the communication gets stuck and nothing appears in the output anymore. If I replace this with a command outputting a shorter string, then it works.

aivarannamaa, 2021-09-21:
I found out that the same issue happens (and same work-around fits) on Lolin D32 pro, which uses the same USB-serial converter. On the other hand -- ESP8266 devices using the same converter are not affected.

This may be related to https://github.com/espressif/esptool/issues/653, because I think I used Lolin D32 pro without problems a few months ago.

aivarannamaa, 2023-05-27:
I now tried firebeetle in fresh Ubuntu 23.04. At first I didn't see the port, but [removing brltty](https://0xsuk.github.io/posts/2022-07-19-how-to-install-ch340-on-ubuntu-22.04/#devttyusb0) helped and I was able to use MicroPython 1.16 and Thonny 4.1 without problems.

All went well with MicroPython 1.20.0 as well.

closed 2023-05-27 by aivarannamaa