thonny/thonny #1965: File not closed

AshDini, 2021-09-12:
**It seems when writing a file on an ESP32, and the file is closed, the Modified time does not get updated until Thonny is closed.**

I am using Thonny 3.3.13 

[testdir.txt](https://github.com/thonny/thonny/files/7149174/testdir.txt)

aivarannamaa, 2021-09-12:
Thank you for the report! I assume you mean right-clicking on the file and selecting "Properties".

Please see whether "Refresh" works for you:

[image]

AshDini, 2021-09-12:
It works if I issue a file.flush (No opening or closing bracket ?)

AshDini, 2021-09-12:
This should not be necessary. When the file is closed, the directory
entries should be updated automatically.

Anyhow flush will achieve that. I just don't know why the brackets are not
needed.

While I am at it, what is the name of the configuration file for Thonny
running under Linux ? I have been unsuccessfully searching for it on my
Raspberry py.

Regards

Bob Stanton

aivarannamaa, 2021-09-12:
> When the file is closed, the directory entries should be updated automatically.

I'm considering this, but it would mean refreshing the list after each run and after each REPL entry. It's not feasible to detect whether a piece of code will change filesystem or not.

> what is the name of the configuration file for Thonny running under Linux ?

It's called configuration.ini on all platforms. The directory can be found via "Tools => Open Thonny data folder"

closed 2021-09-14 by aivarannamaa