thonny/thonny #1968: external modification confirmation

kr-g, 2021-09-14:
this is not a bug.

when running blake many files get manipulated external... 
i m sorry ... somehow ... my coding style is really that bad.

in thonny it is required to confirm each file for being reloaded.

would be nice to have a "yes to all" button

it s not of a high prio to me. but would be nice.

aivarannamaa, 2021-09-14:
Thank you for the suggestion!

I'm leaving it in the TODO list.

kr-g, 2022-08-05:
one note onto this:

currently all is triggered by `EditorNotebook.on_focus_window()` but in case a script is executed and manipulates another file (e.g. a log or txt file what is open in another editor) there is no such event coming and thonny is unaware of a changed file.

i think there should be running another periodically task (e.g. on workbench => some tk_widget.after(delay, callback=None)) what will check every 5sec (?)

RFC

aivarannamaa, 2022-08-09:
> in case a script is executed and manipulates another file (e.g. a log or txt file what is open in another editor) there is no such event coming and thonny is unaware of a changed file.

Good point!

> i think there should be running another periodically task

It's would be more efficient to make a check each time the user has executed something in Thonny. You can register a listener for this event like this: `get_workbench().bind("ToplevelResponse", do_stuff, True)`

kr-g, 2022-08-14:
listener added

aivarannamaa, 2022-08-14:
I'm sorry, I had forgotten my earlier plan how to deal with external modifications #1589.

As you can see from the last commit under this issue, I solved it in a simpler way -- if the file was saved in Thonny (ie. not "dirty") before the external modification, the reload happens automatically, without any confirmation. This matches the behavior of VSCode and PyCharm.

For each dirty editor or editor with deleted underlying file, there will be separate confirmation dialog, as before.

I'm still grateful for your pull-request, because it pushed me to solve it earlier than I had planned :)

closed 2022-08-14 by aivarannamaa

kr-g, 2022-08-14:
just wondering why other (also *unix) tools ask here for an extra confirmation. 
probably because of preventing reloading continuously frequently changed files (like logging) and consuming therefore cpu?
mhmmm....
whatever.

kr-g, 2022-08-14:
dirty editors are confusing anyway...

https://github.com/thonny/thonny/issues/1041

somehow i feel that all a bit complicated / "mixed"

aivarannamaa, 2022-08-14:
> somehow i feel that all is a bit complicated / "mixed"

What do you mean?

kr-g, 2022-08-14:
it is just a "feeling"...
as i wrote.
that is different from "thinking"
sorry but i can not explain this in a better way...

kr-g, 2022-08-14:
but when vscode and pycharm do it like that... there might be a good reason for (which i m unaware...)