thonny/thonny #1970: RPI pico not recognised in Windows 8

ESCcrascirepository, 2021-09-21:
The RPI Pico flashed with MicroPython Ver1.17 is not being recognised by Thonny in Windows 8. The same board flashed with the same MicroPython Version is being recognised in Windows 10.  The same applies for circuit Python also.
The computer understands that the device plugged in is a micro-python peripheral device in both Windows versions. But In Windows 8, it is not recognised in Thonny.  
Screenshots of the same device in both windows versions running Circuit and Micro Python are attached:

**Specs:**
PC: WIN 8
Micro Python Ver: 1.17
Circuit Python: 6.3.0
Flash Nuke: Attempted Twice
Thonny Ver: 3.3.13
Anti-Virus: All access has been given to thonny for accessing the USB peripherals.

_Win 10- Micro Python_
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w10-mp.png)
_Win 10-Circuit Python_
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w10-cp1.png)
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w10-cp2.png)

**_### ERRORS ARE PRESENT IN WIN8:_**

_Win 8- Micro Python_
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w8-mp.png)
_Win 8-Circuit Python_
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w8-cp.png)
[IMAGE](https://www.esccrasci.in/MicroPython/Errors/w8-cp2.png)

**Video Proofs**
WIN 8:
[CP](https://vimeo.com/610875108)
[MP](https://vimeo.com/610874979)
WIN 10: Video proofs not supplied due to the functional status of thonny in Win 10.

If this error is fixed, it will be great since Mu is also facing similar issues in Windows 8. 
Began as a thread [here](https://twitter.com/ECcrasci/status/1439293319754903554)

aivarannamaa, 2021-09-21:
Thank you for the comprehensive report!

Please check the "Ports (COM & LTP)" section in the Device Manager of your Win 8. Which COM ports do you see (if any) ?

aivarannamaa, 2021-09-21:
You may also want to check this thread: https://github.com/raspberrypi/pico-feedback/issues/118

ESCcrascirepository, 2021-09-21:
Thanks a lot for sharing the link to the other repo. The solution there has fixed the issue. Thank You!

aivarannamaa, 2021-09-22:
Great!

closed 2021-09-22 by aivarannamaa