thonny/thonny #1972: Theme changes back when opening thonny

Aandrine, 2021-09-22:
I am using a MacBook Air M1, 2020, version 11.3.1, and I have Thonny version 3.3.13 installed. I have used thonny previously on another computer and been very happy with it, especially because I can customize how it looks. I am a big fan of dark mode, and the first thing I did after installing thonny on my new mac was to go and change the theme to dark mode. However I quickly noticed that it changes back into light mode every time I reopen the program. This did not happen on my last computer, and it's honestly so annoying that I might quit using thonny if I cant stop it from happening. Does anyone know how I can "permanently" change the theme to dark mode?

aivarannamaa, 2021-09-22:
Thank you for the report!

Do any other changes to the settings stick? For example, when you resize Thonny's window and close and reopen it -- does the changed size stick?

aivarannamaa, 2021-11-06:
Please re-open if you have more details to add?

closed 2021-11-06 by aivarannamaa