thonny/thonny #1973: Thonny 3.3.14 Device is busy or does not respond

Joe-Jefferies, 2021-09-23:
I started using Thonny to program RPI Pico in MicroPython yesterday and everything was fine - it found both my existing Python 3.x on my PC and the newly installed Pico on USB.

Without changing anything I now get the above ( Device is busy ) message
I tried:-
Reboot PC
Uninstall Thonny and re- install
Re-install firmware on Pico
Tried another Pico Board
Disconnect Pico 

All to no avail
Please can anyone help
Joe

aivarannamaa, 2021-09-23:
Thank you for the report!

Did you save anything into `main.py`? A loop perhaps? If yes, then did you try exiting it with Ctrl+C or Stop/Restart?

aivarannamaa, 2021-09-23:
Also, can you connect to your device with a serial terminal program (eg. Putty or tio)?

Joe-Jefferies, 2021-09-23:
Thank you for your reply

There is no Main.py and the problem exists even without pico connected

Joe

respond (#1973)

<https://github.com/thonny/thonny/issues/1973#issuecomment-925725714> , 
or Android

aivarannamaa, 2021-09-23:
> the problem exists even without pico connected

Do you mean that Thonny is giving you "Device is busy or does not respond" without Pico connected? In this case you must be connecting to a wrong port. 

Please select "Run => Select interpter => MicroPython (Raspberry Pi Pico)" and tell me, what are the options in the "Port" dropdown.

Joe-Jefferies, 2021-09-24:
Hi again

I have nailed the problem - the pico had been working previously so I 
had swapped the pico for another brand new one but got same problem
I swapped the board but not the USB cable.  The cable must have suffered 
some damage in its life and now works intermittently. I have now 
replaced USB with a new one and all is fine

Thank you very much for your help in this - I am new to Pico and 
MicroPython/Thonny so missed a basic hardware issue
Joe

respond (#1973)

<https://github.com/thonny/thonny/issues/1973#issuecomment-926069626> , 
or Android

aivarannamaa, 2021-09-24:
Wish you luck with getting to know MicroPython!

closed 2021-09-24 by aivarannamaa

qkum, 2022-04-23:
> Thank you for the report!
> 
> Did you save anything into `main.py`? A loop perhaps? If yes, then did you try exiting it with Ctrl+C or Stop/Restart?

`Ctrl+C or Stop/Restart?`

_WHERE?_

On the computer working with the Pico?  (if so, where? on the Desktop screen?)
Inside the Thonny IDE?
Inside the Thonny Shell?
Inside the Pico Storage Folder? (in ROOTSEL mode)

- Extremely useless error message is my point.
- Because it is just the default error message it seems. 
- Following it does not help a tiny bit if the solution in reality is re-flashing the Pico.
- Using Thonny to re-flash (update firmware) my Pico did not change anything. It "worked", and it rebooted, but still same problem afterwards and able to change the script on the pico.

Google lead me here, and the broken "error message" is the reason I choose to dump in on a old post and write this. 

I hope my input can lead to a more useful error message in the future. 
- When you are new to stuff, the error message has to be more in-depth that this, to be of any use, is my point.

Have a nice day and thanks for your work on this project 💯