thonny/thonny #1975: [suggestion] column mode editing

atesin, 2021-09-24:
as identation and vertical alignment is SO important in python, i think this feature would be extremely useful

to trying to convince you, look at this example with notepad++ and imagine doing so in thonny

[image]

aivarannamaa, 2021-09-24:
Thank you for the clear message!

In Thonny you can change indentation by selecting the rows and pressing Tab or Shift+Tab. Wouldn't this meet your requirements?

atesin, 2021-09-24:
hi ... thanks for answer early
yes it also could be done with (de)indent ... but this feature will go further :)

as a sample, i updated the initial post with a new gif covering array editing... take a look

aivarannamaa, 2021-11-04:
Sorry for the delayed reply!

Thanks for the gif, but I'm afraid it would be too hard to implement on Thonny's UI framework (Tk). Even displaying multiple cursors (or a multiline cursor) would need hard hacking.

closed 2021-11-04 by aivarannamaa

Zahu2018, 2023-01-28:
# A simply - stupid multicursor only for insert text in tkinter (if needed) :))
import tkinter as tk

positions = []  # list of cursor positions of cursor
#                 when 'Multicursor' button is red
te = """
            HOW IT WORK

1. Click 'Multicursor' button to begin session.
2. 'Multicursor' button is red.
3. With Left Button of mouse make selections.
4. When selection is ok add desired text in entry box.
5. Click 'Multicursor' again to insert new text.

(Note: When selection is activate, 'Multicursor' button is red;
       otherwise is black.)
"""
    
def handler2(event):
    coord = t.index(tk.INSERT)
    positions.append(coord)
    print(coord)
    print(positions)
    
def scrie():
    b.config(command=multicursor)
    b.config(fg='black')
    new_text = e.get()
    print(new_text)
    for position in positions:
        t.insert(position, new_text)
    positions.clear()
    e.delete(0, 'end')
    
def multicursor():
    b.config(command=scrie)
    b.config(fg='red')
    positions.clear()
    t.bind('<ButtonRelease-1>', handler2)    

window = tk.Tk()
window.title("Multi Cursor Insert Text")
t = tk.Text(window)
t.grid(row=0, columnspan=3)
t.focus()
t.insert(1.0, te)
b = tk.Button(window, text="Multicursor", command=multicursor)
b.grid(row=1, column=0, sticky='e')
e = tk.Entry(window)
e.grid(row=1, column=1, sticky='w')
window.mainloop()

atesin, 2023-02-04:
hi @aivarannamaa   ... i got an email with a comment is not shown here, with a suggestion and a snipped code on how to achieve this... did you read it?, do you delete it? ....  by making an array with cursor positions

aivarannamaa, 2023-02-10:
@atesin, I haven't seen this. I suppose the author of the comment deleted it later.