thonny/thonny #1976: [suggestion] jump to function / cursor position history

atesin, 2021-09-24:
hi... thanks for thonny, we are making it better together
this proposed feature will make code navigation a lot easier:

in some other editors you can ctrl+click a function call (like a hyperlink) and the cursor/line/scroll/tab jumps to the function definition, this is useful to lookup a function or object and remember their workings and structure

then additionaly you can go back your previous scroll position(s) with a "back" button or hotkey and continue coding

aivarannamaa, 2021-09-24:
Thank you for the suggestion!

Ctrl+Click should already work, but without any visual affordances (eg. changed cursor), though. I will enhance this in 4.0

Moving back and forth with Alt-arrows would be a good addition. I'll keep this issue for reminding this to myself.

atesin, 2021-09-24:
also you could reserve arrows for future use in code completion features (like bash/doskey command history) and do "back" with ctrl+right click, ctrl+backspace, F12, Fx or whatever

just ideas