thonny/thonny #1977: [bug] thonny reopens the current file when try to open another instance with other file

atesin, 2021-09-24:
description:
- i have thonny opened with the file first-file.py
- i open another thonny instance with the file new-file.py (by command line or explorer context menu (see below) )
  #1962 
- thonny windows fickers (like if were quickly reopened) and open the same current first file
- with single instance enabled, it flickers and nothing seem to change
- with multiple instance, it flickers and reopen 2 windows, both with first-file.py... and 3, 4, N windows with same file
- if i open the file inside thonny with menu > file > open, it opens the file in a new tab as expected
- additionally, seems thonny don't support drag+drop files to window to open (suggestion?)

environment:
- windows 8.1 pro x64 spanish
- python 3.9.7
- thonny 3.3.14 installed with 'pip install thonnyapp'
- intel xeon 2.4ghz, 12 cores, 12gb ram, 1tb hd half full, nvidia gtx1650 4gb

atesin, 2021-09-24:
update:
- by right clicking the file and use explorer context menu as installed in tutorial, it shows the behavior described in the bug
- with windows console (cmd.exe) it it flickers, but the file finally is opened in a new tab/window
- maybe an additional step/argument is needed in context menu to work properly (btw what means "-m thonny"?, does thonny have command line arguments?)

atesin, 2021-09-28:
hi... me again
until this is not so much related to thonny base code itself and is more related a 'mod' i had done... anyway could be useful to be aware of this for the moment @aivarannamaa will write an official thonny windows explorer shell extension [as he said](https://github.com/thonny/thonny/issues/1962#issuecomment-920655740) ,  and resolve the problems i had, previous to release

aivarannamaa, 2021-11-04:
Thank you for the report!

I've never tried command line arguments with multiple instance mode. I'll try to reproduce and solve it later.

atesin, 2021-11-05:
i suspect it has to do with launching thonny with different user accounts...

if i open a cmd console and run> thonny.exe C:\dir\myfile.py  ... indeed it opens/reutilize the thonny instance with the file opened in a tab....  but maybe the the right click context menu/regedit runs with some system account and the permissions blocks the parameters/files/instances somewhere