thonny/thonny #1978: [question] assistant: indentation warnings are driving me crazy, how to disable them?

atesin, 2021-09-25:
my assistant log are flooding with indentation warnings... i always use 2 spaces instead 4 for indents to prevent my code gets too
 W I D E

in the meantime this (hidden) option is not available, i would like to disable related warnings, and learn **how to use the assistant** at the same time, because i couldn't find any documentation about it

thanks my prays are listened

aivarannamaa, 2021-09-28:
Please go to "Tools => Options => Assistant", and paste `bad-indentation` to the big box.

aivarannamaa, 2021-09-28:
You can check other Pylint message codes here: https://github.com/janjur/readable-pylint-messages

aivarannamaa, 2021-11-04:
Hope my recommendation solved your problem

atesin, 2021-11-05:
sorry to not had replied... surely it did, thanks for your answer
... could you add this link to thonny help file?

aivarannamaa, 2021-11-05:
Added it to wiki: https://github.com/thonny/thonny/wiki/Assistant. Will create internal help page for Assistant later.

atesin, 2021-11-05:
thanks...  sure this could help more people besides me

i was taking note some other ideas i will open suggestions now

closed 2022-05-14 by aivarannamaa