thonny/thonny #1979: thonny is not in the official repros op ubuntu 21.10

speedy-10, 2021-09-25:
Hi, I just can't install thonny by `sudo apt  install thonny` on 21.10. 
I know other installing methods, I just want to share the observation.

Regards

Polosssatik, 2021-09-27:
Hello, can you tell about other methods?

aivarannamaa, 2021-09-28:
@speedy-10, thank you for the notification! I asked for advice here: https://askubuntu.com/questions/1366211/how-to-find-out-why-a-debian-testing-package-is-missing-from-ubuntu

@Polosssatik, please see https://github.com/thonny/thonny/wiki/Linux

TechCowboy, 2021-10-22:
This works on 21.10
bash <(wget -O - https://thonny.org/installer-for-linux)

aivarannamaa, 2021-11-06:
Created a bug-report at Ubuntu: https://bugs.launchpad.net/ubuntu/+source/thonny/+bug/1950037

aivarannamaa, 2021-11-07:
@TechCowboy, @speedy-10, @Polossatik, I created bug report at Ubuntu: 
https://bugs.launchpad.net/ubuntu/+source/thonny/+bug/1950037

You can raise the priority of this bug by marking that it affects you and/or subscribing to this bug.

speedy-10, 2022-02-05:
I just tested the installation on ubuntu 22.04: thonny 3.3.14 can be installed by sudo apt install thonny, nice.

closed 2022-02-05 by speedy-10