thonny/thonny #1980: Thonny detects Pico the first time during setup but stops detecting after that

DominatorCruz, 2021-09-26:
Hello, having trouble connecting Thonny to Pico. Followed https://core-electronics.com.au/tutoria ... honny.html and was perfectly fine. But when I tried plugging the pico a second time Thonny no longer detects my Pico. Have written only one program on my Pico 
`from machine import Pin
from time import sleep
led = Pin(25, Pin.OUT)
n = 0

while True:
    led.toggle()
    print("13 x {} = {}".format(n, 13*n)) # print the thirteen-times table
    n = n+1
    sleep(0.5)`

Have tried clicking the stop button & pressing Ctrl + C to end program , but when I would try to access files in the Raspberry Pico, this dialog would pop up [https://ibb.co/VT29HVF](url)
Have tried pressing BOOSET & updating firmware.
Am using Micropython 1.17 (sysname='rp2', nodename='rp2', release='1.17.0', version='v1.17 on 2021-09-02 (GNU 10.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040') 
Running the Thonny IDE on Windows 10

aivarannamaa, 2021-09-28:
Thank you for the report!

There seems to be a problem with MicroPython 1.17 (https://github.com/micropython/micropython/issues/7867). Please check, whether 1.16 works for you.

DominatorCruz, 2021-10-02:
1.16 works fine

aivarannamaa, 2021-10-02:
@DominatorCruz, that's good to know! If would be really helpful, if you tried to replicate the experiment mentioned under https://github.com/micropython/micropython/issues/7867 with a serial terminal (eg. `tio` under Linux or Putty under Windows) and reported your results. If you can also confirm that it's a regression, then the MicroPython developers would probably take it more seriously.

DominatorCruz, 2021-10-05:
With micropython 1.16 when I first press Ctrl+A and then Ctrl+C the input gets stuck but it works fine when I press Ctrl+D and then Ctrl+C. With micropython 1.17 is is a regression of both scenarios having problems.

aivarannamaa, 2022-04-09:
Fixed in https://github.com/micropython/micropython/commit/587339022689187a1acbccc1d0e2425a67385ff7 and current nightly build seems to work nicely: https://micropython.org/resources/firmware/rp2-pico-20220409-unstable-v1.18-307-g66b5c4c77.uf2

closed 2022-04-09 by aivarannamaa