thonny/thonny #1981: CANNOT install Package TENSORFLOW in THONNY IDE 3.3.13

rosarion, 2021-09-26:
I got Errors after making THONNY to run a PYTHON 64-BIT version  as per advise..
The PYTHON.exe 64-bit is not compatible with THONNY 
Traceback (most recent call last):
  File "C:\Thonny\lib\runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\Thonny\lib\runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "C:\Thonny\lib\runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "C:\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 5, in <module>
    import tkinter as tk
  File "C:\Thonny\lib\tkinter\__init__.py", line 36, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
ImportError: DLL load failed: %1 is not a valid Win32 application.

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

rosarion, 2021-09-26:
[image]

rosarion, 2021-09-26:
Here is a working THONNY in WINDOWS 1
[image]
0

aivarannamaa, 2021-09-28:
How did you precisely made Thonny to run Python 64-BIT version?

From your first screenshot, it looks like Thonny is using the bundled Python. Did you change anything in Thonny's installation directory?

aivarannamaa, 2021-11-06:
Please re-open if you have more details to add!

closed 2021-11-06 by aivarannamaa