thonny/thonny #1985: Error updating connectivity to pico

timber101, 2021-09-30:
Hello, I'm trying to connect Thonny to my pico but get this error..

What do I need to do.

Thanks

Colin

---------------------------
Could not determine download url
---------------------------
Could not find the right file from the release info (https://raw.githubusercontent.com/thonny/thonny/master/data/rpi-pico-firmware.json)
---------------------------
OK   
---------------------------

aivarannamaa, 2021-09-30:
Thank you for the report!

You got this message while trying to install MicroPython for your Pico, right?

When you plug-in your Pico while holding BOOTSEL button, you should see a drive with a file named INFO_UF2.TXT on it. Could you please give me the content of this file?

timber101, 2021-09-30:
Thanks

Here it is

UF2 Bootloader v1.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2

timber101, 2021-10-02:
Hi There, it was my mistake, I downloaded the nuke file and then expected it to work.

have downloaded the correct uf file am coding

Sorry for any inconvenience!

Colin

timber101, 2021-10-02:
as above, my bad, didn't download the right uf file

closed 2021-10-02 by timber101