thonny/thonny #1986: Installing package micropython-ssd1306 leads to error: SSL: CERTIFICATE_VERIFY_FAILED, certificate has expired

Jan-Rekers, 2021-10-03:
I am new to user Thonny. I have installed version 3.3.13 on a Windows 10 machine, Windows is fully up to date.
On using Tools/Manage Packages I search for SSD1306 and get suggested the package micorpython-ssd1306 which is exactly what I was looking for.
Homepage: https://github.com/stlehmann/micropython-ssd1306
PyPI page: https://pypi.org/project/micropython-ssd1306/

However, on choosing this package I immediately get this error message:
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)>

I had a similar problem when I tried to switch Thonny to MicroPython (Raspberry Pi Pico). I could avoid that problem by dropping the file rp2-pico-20210902-v1.17.uf2 directly on the F: drive.

Is there a similar solution for the Install Package problem?

Thanks, Jan Rekers

Jan-Rekers, 2021-10-03:
I have managed to circumvent the above problem in the following way:
- I went to the GitHub page of the ssd1306 package: https://github.com/stlehmann/micropython-ssd1306
- I found the file ssd1306.py, copied all python code to the clipboard
- I pasted this code in a new file in Thonny
- On choosing RUN, I choose to install on the Raspberry Pi Pico, under the name ssd1306.py
- Now, in my own code the following line does not give a runtime error anymore: from ssd1306 import SSD1306_I2C

Hope this helps anyone facing the same problem.

aivarannamaa, 2021-10-03:
Thank you for the report!

I can reproduce this in one of my Windows machines but not on another and not in Ubuntu.

Related discussion on MP forum: https://forum.micropython.org/viewtopic.php?f=15&t=11201

Another, possibly related discussion: https://borncity.com/win/2021/09/30/sept-30-2021-will-we-see-trouble-with-old-lets-encrypt-certificates/

Can you please check, whether your browser is happy with certificates of https://micropython.org/ (for example, in Chrome or Firefox click on the padlock left of the address bar). If you ask for detailed information, do you notice some warnings?

aivarannamaa, 2021-10-03:
Another, possibly relevant issue: https://github.com/certifi/python-certifi/pull/162

Jan-Rekers, 2021-10-03:
As it turns out, Chrome considers the certificate of https://micropython.org/ as VALID, however, Firefox does not...!
Firefox: SEC_ERROR_EXPIRED_ISSUER_CERTIFICATE.
According to Chrome the issuer of the certificate is Bitdefender Personal CA.Net-Defender, which it trusts.

evgeni, 2021-10-03:
@Jan-Rekers it seems you have a proxy/middlebox that re-encrypts your internet traffic? micropython.org is signed by Let's Encrypt: https://crt.sh/?q=micropython.org

Jan-Rekers, 2021-10-03:
@evgeni: I do use Bit Defender which might function as a proxy in Chrome. I am surprised that Chrome states that BitDefender did issue the certificate of MicroPython.org, while that is not the case. Still, it does.
I have added an exception in Firefox to trust the certificate of MicroPython.org, even though it is signed by Lets Encrypt. Now Firefox will open the site.
Still, that does not solve the problem in the Thonny aplication: it still considers the certificate of MicroPython.org as not-trustable as it does not trust LetsEncrypt.

@aivarannamaa : is there a way to tell Thonny to trust LetsEncrypt as certificate issuer? Of would it be possible for MicroPython.org to use a real certificate authority instead of LetsEncrypt?

Thanks for all the help up to now!

Kind regards, Jan Rekers

aivarannamaa, 2021-10-05:
No, Thonny don't have means for tweaking the SSL operation. It must be fixed on OS side: https://bugs.python.org/issue45372

I'm at loss here. I tried suggestions given at https://community.letsencrypt.org/t/fixing-windows-installs-that-dont-receive-updates-to-their-trusted-roots/161162/28 but it didn't work for me. It looks like we're not the only ones who can't pinpoint the reason why some Windows instances are not properly updating their certificate stores.

aivarannamaa, 2021-10-05:
I finally got it solved by installing https://letsencrypt.org/certs/lets-encrypt-r3.der

Jan-Rekers, 2021-10-06:
That has solved my problem! Thonny is now able to receive packages from micropython.org.

Still, I do not really like the fact that I had to manually update the trusted root certificates on my computer because "someone on the internet suggested it"... I would have preferred a solution where windows itself would have updated its trusted root certificates. That is a windows issue, though.

Many thanks for the extensive help!

closed 2021-10-06 by Jan-Rekers

T-Wainwright, 2022-02-15:
Hi All,

Is there a permanent fix for this on the horizon at anypoint? We use Thonny in a teaching lab, and are having multiple students running into this same issue with personal windows devices. As @Jan-Rekers pointed out manually downloading and updating a trusted certificate isn't really a fix, more of a hack with a number of potential security questions attached. 

All the best,

Tom.

aivarannamaa, 2022-02-15:
> Is there a permanent fix for this on the horizon at anypoint?

I don't know how to fix it in Thonny as the problem is at the OS level. As a work-around I could switch off certification verification for micropython.org requests, but this isn't too elegant either.

aivarannamaa, 2022-02-15:
... I also considered switching to using certifi-s certificates, but the same problem occurs there as well: https://github.com/certifi/python-certifi/pull/162

BongerBob, 2022-03-29:
Can confirm, this fixed for me ( as listed above by another user ) : ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1091)

Windows 10, chrome for browser, updates are current as at 29/03/2022.

emperinter, 2022-03-30:
It Works For me !

stokes776, 2022-07-03:
How exactly are you guys 'fixing' this? I downloaded the Cert and loaded it into both my computers Personal cert store, as well as into Firefox and neither worked, still getting the error.

peppe8o, 2022-07-23:
Hi all. In my Windows 10 PC, I was able to fix the error by running Thonny with Administrator privileges. After that, I was able to download the MicroPython firmware without errors.

MatGrump, 2023-01-24:
I can confirm I have this issue on windows 10. How about allowing users to bypass SSL errors with a tick box or something?

aivarannamaa, 2023-01-26:
@MatGrump, I'm planning to allow users to use alternative certificates (#2341), bundled with Thonny.

budley, 2023-07-30:
Maybe this issue shouldn't have been closed?
Newcomer to Thonny here.
Tools | Manage Packages
Search (for anything)
"Could not fetch search results: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1007)"

randyodom, 2023-08-26:
> I finally got it solved by installing https://letsencrypt.org/certs/lets-encrypt-r3.der

I have been trying to solve this issue for a couple of days. I found this, and it solved the problem. I can now add packages to Thonny again. I just signed up to GitHub to post my thanks.

TechCowboy, 2023-10-16:
It seems this has risen from the dead.  Oct 16, 2023

[image]

aivarannamaa, 2023-10-17:
@TechCowboy, this error message differs from the one in the title of this thread. Which Thonny version are you using? Have you been able to install packages with the same Thonny installation before?

TechCowboy, 2023-10-17:
I received this error with 4.1.3, and an earlier version (4.1.2?).  I've
used Thonny for years and have no problems installing packages in the
past.  This is with Windows 11.

The latest version available on Ubuntu is 3.3.14-1 and works fine.

TechCowboy, 2023-10-17:
The Windows 11 was a fresh installation (format and clean install), so I
had not installed packages on this "new" machine before.

proximacentaury, 2024-03-29:
The Thonny with administration right and install of the certificates worked for me...

cturpel, 2024-08-15:
on OSX go to finder -> python3.x folder in applications and run the
[image]

Restart Thonny and ready to go.