thonny/thonny #1988: visual debugging and shell auto scroll

kr-g, 2021-10-04:
when debugging visual the dynamic output in the shell tab the content scrolls away to the last line in the textcontrol.
it would be very helpfull to have sort of logic (or just a checkbox beside the tab caption) to stop the autoscroll feature.

aivarannamaa, 2021-10-11:
Thank you for the suggestion! Can you please give me an example workflow which is hindered by current behavior? Maybe I can suggest a work-around.

kr-g, 2021-10-11:
ok well ... following the output is a challenge in case i run a python program which executes a loop what processes regular based incoming data (especially with micropython). 

one quick win here would be to write the shell output also to a extra logfile, so someone could use "tail -f logfile", or "less".

workaround... if you have a better idea... an option would be to close the connection to the device (with each start) and use a shell utility to grab the output coming in from /dev/ttyUSB0. 

but this would break somehow the idea of an IDE (because not "I"ntegrated, just beeing a "D"evelopment "E"nvironment)

aivarannamaa, 2021-10-13:
Fair enough. As it's not difficult to implement, I'll queue this for 4.0. I'd like to keep the UI as clean as possible for newcomers, though, so I'm thinking about putting the option inside shell's context menu. Would this be OK with you?

aivarannamaa, 2023-07-07:
Another idea -- only keep scrolling when the bottom line is visible. Stop scrolling when the user has already moved upwards.

kr-g, 2023-07-07:
fine with me

closed 2023-12-02 by aivarannamaa