thonny/thonny #1990: Importing and Using CV or OpenCV to Thonny

ImprovedManX, 2021-10-06:
Hello, I am attempting to download cv or opencv to my Thonny environment. I am able to install the file (I think). I have tried moving the file into the Thonny lib folder and many other folders with no avail. Every time I try to import the file it says the file does not exist, is there any way I can use opencv on Thonny? (On a Chromebook if that changes anything)

r0the, 2021-10-13:
The name of an opencv [Python package](https://pypi.org/project/opencv-python/) is `opencv-python`.

ImprovedManX, 2021-10-13:
cv2 Worked, thank you for the help.

closed 2021-10-13 by ImprovedManX

sleezy, 2023-03-30:
> The standard way of installing packages is with pip (Thonny's "Tools => Manage packages" uses pip). Have you tried it?

please that's what I get trying to do that .

even though I have no issues when I use python idle , its succefull

Successfully installed numpy-1.24.2 opencv-python-4.7.0.72
Starting to apply changes to the target.
Copying opencv_pythonError Could not write next block after having written 0 bytes to /lib/cv2/.dylibs/libSvtAv1Enc.1.4.1.dylib

aivarannamaa, 2023-03-30:
@sleezy are you trying to install `opencv-python` for a MicroPython board? I'm afraid this is not possible, because this package is meant for CPython.

aivarannamaa, 2023-03-30:
@sleezy, are you currently using your Thonny for CPython or MicroPython development? You can check this from the lower-right corner of Thonny's main window.

sleezy, 2023-03-30:
> @sleezy, are you currently using your Thonny for CPython or MicroPython development? You can check this from the lower-right corner of Thonny's main window.

Micro python

sleezy, 2023-03-30:
Micropython development 

> @sleezy, are you currently using your Thonny for CPython or MicroPython development? You can check this from the lower-right corner of Thonny's main window.

aivarannamaa, 2023-03-30:
> Micropython development

If you want to use OpenCV then you need to switch from MicroPython to Python 3.

sleezy, 2023-03-30:
Ooh really.. so how do I do that  on thonny .. thanks

sleezy, 2023-03-30:
> Ooh really.. so how do I do that on thonny .. thanks

I have python3 installed on my Mac already .. was trying to run a class attendance program using rfid n face recognition but on using the python idle mac keeps blocking the use of camera . So I checked esp32 manual n I saw a sample they used thonny and the camera worked . So now trying to run my own code with the same micropyton was giving me cv2 errors

Re-Kunz, 2023-09-12:
Tools > Manage packages... > "opencv-python" > Search on PyPI = Search results : klick on "opencv-python" > Install :)