thonny/thonny #2006: Tonny will not load

AshDini, 2021-10-19:
After reinstalling Thonny (on win10 with Python 3.9.6 already installed ), will not allow me to use Esp32 or 8266.

When I try under Tools/Options to switch to ESP32 I get an internal error "ModuleNotFoundError: No module named 'serial.tools'"

Tried 3 times, always the same result !

here is the Copy of the Internal error dialog:

**Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 122, in _backend_changed
    page = self._get_conf_page(backend_desc)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 141, in _get_conf_page
    self._conf_pages[backend_desc] = cp_constructor(self.labelframe)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\esp\__init__.py", line 126, in __init__
    super().__init__(master, "esp8266", "0x0")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\esp\__init__.py", line 80, in __init__
    super().__init__(master)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 422, in __init__
    for p in list_serial_ports()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 880, in list_serial_ports
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'**

Regards
Bob Stanton
bobstantonx@gmail.com

closed 2021-10-20 by aivarannamaa

AshDini, 2021-11-10:
> Thank you for the kind words!
>
> Unfortunately MicroPython doesn't have required technical features for
> building a (useful) debugger for it. There is something, but not enough and
> even this would require different MicroPython build.

Hello Aivar,

There are a couple of small errors in Thonny:

1) flush() - gives 'ValueError:'
    flush - without brackets is OK

2) When deleting a file from an esp32 using Thonny, after the deletion,
pressing OK will bring up an "Internal error" dialog. CANCEL does not.

And a question: Could you give me some help to create some "Compiler
Directives". Such as: Clear the shell after the run, or do not save before
each run, etc.
Such facilities, if possible to implement, would be really helpful in
testing the code.

Regards

Bob Stanton

aivarannamaa, 2021-11-10:
1) flush without brackets doesn't do anything. You need to investigate whether the object supports flushing in given state. I believe this is not specific to Thonny

2) Need more details. Please open another issue

The extra features you mentioned have not been implemented. I added one of them into TODO list (https://github.com/thonny/thonny/issues/2039). Some of them were planned for 4.1 before.

AshDini, 2021-11-12:
Hello Avair,

Here is a step by step description how the error happened (iwas in the
ESP32 mode, trying to delete a file)

1) pressed "File" in the menu - displayed "where to open from"
2) Clicked on "Micropython device" - displayed list of files
3) Right clicked on file to be deleted - displayed submenu
4) Clicked on "Delete" - displayed "Are you sure" dialog
5) Clicked on "Yes"
6) File is deleted
7) Clicked on OK - displayed a dialog headed "Loading ( see image 3)
    Strictly speaking, the "OK" button is really superfluous. The dialog
showing all the files should just stay there
    (in case of the user wants further deletions, after which the "Cancel"
should close the dialog (just as it does now)
8)  Clicked Close - Internal error dialog displayed with the following
content:
-----------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705,
in __call__
    return self.func(*args)
  File "C:\Program Files
(x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1961, in
toolbar_handler
    handler(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 781, in _cmd_open_file
    self.show_file(path, propose_dialog=False)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 926, in show_file
    editor = self.get_editor(filename, True)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 1015, in get_editor
    return self._open_file(filename)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 999, in _open_file
    if editor._load_file(filename):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 169, in _load_file
    result = self._load_remote_file(filename)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\editors.py",
line 228, in _load_remote_file
    raise RuntimeError(response["error"])
RuntimeError: Problem with a management command

SCRIPT:
__thonny_fp = __thonny_helper.builtins.open('/main.py', 'rb')

STDOUT:

STDERR:
Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

OSError: [Errno 2] ENOENT
---------------------------------------------------------------------------------------------------------------------

Hopefully, that fully describes how this (minor) error occurs. Of course,
it is easily overcome just by pressing "Cancel" instead of OK in Step five.

By the way, I am using Thonny 3.3.23 and the esp32-20210902-v1.17 version
of Micropython.

Regards

Bob Stanton

P.S. You mentioned that "The extra features you mentioned have not been
implemented. I added one of them into TODO list (#2039
<https://github.com/thonny/thonny/issues/2039>
<https://github.com/thonny/thonny/issues/2039>). Some of them were planned
for 4.1 before."

Does that mean that there is a 4.0.0. version and when will it be available
?

aivarannamaa, 2021-11-12:
I see that you have tried to delete a file via file *opening* dialog. I was not aware that this is possible. I'll fix this.

File deletion is supposed to be invoked via the file browser (View => Files).

AshDini, 2022-02-07:
Hello Aivar,

This is just a small observation, and I don't even know if I am correct
about it, or just being too particular.

I think that in the case where there are Libraries in the ESP32
directories, the files are indented, and appear as if they are under the
lib folder.

See attached example "0.jpg":
I think it should look like "4jpg"

I thought of this as I was writing a recursive directory list.

If you think that I am wrong or too pedantic, please ignore this email.

Regards

Bob Stanton

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
02/08/22,
08:38:38 AM

aivarannamaa, 2022-03-23:
Sorry for the late reply!

Unfortunately your images didn't come through