thonny/thonny #2007: Acces console to esp32 not working anymore

pidou46, 2021-10-19:
Hello some trouble recently with Thonny and ESP32

Here is my setup:

WEMOS Lite V1.0.0
Thonny 3.3.14 (installed with pip)
Ubuntu 20.04.03 LTS (64-bit)

This setup use to work, until some point but failed now.

Tried with mp V1.16 and mp V1.17

When I switch to ESP32 (/dev/ttyUSB0) it failed now. No error message is printed in console, neither is the terminal I use to launch Thonny.

There is no custom file on esp32 fs only the default boot.py

I can aces ESP32 using:
 picocom /dev/ttyUSB0 --baud=115200

I'm also able to upload successfully the firmware using Thonny.

How could I get debug information ? log file ?

Thanks

pidou46, 2021-10-19:
Note : to connect via picocom I need to reset the ESP32 device

pidou46, 2021-10-19:
I don't see anything special, do you ?

aivarannamaa, 2021-10-20:
It looks similar to #1964. You said that the problem occurred overnight. I suspect that you updated your Linux and some change in the USB drivers triggered the problem. Please try this work-around:  https://github.com/thonny/thonny/issues/1964#issuecomment-923862258

pidou46, 2021-10-21:
Thanks I have tried the work-around (options in configuration.ini) but I don't see any change.

I will try with another device as soon as possible.

pidou46, 2021-10-30:
The problem seem's to be linked to Ubuntu CH340 USB serial driver, because it works under windows but not with Ubuntu

CP2102 USB Serial is ok.

The problem affect Thonny but not picocom

pidou46, 2021-11-12:
The problem have gone, probably solved by ubuntu update of driver ?

closed 2021-11-12 by pidou46