thonny/thonny #2025: Missing ">"

Tovim, 2021-10-30:
Hi,
I would like to comment the command "C:\Python37-32\Scripts\pip install thonnyapp" in the  https://github.com/thonny/thonny/wiki/Windows > 'Installing Thonny and Python separately' paragraph. Me seems that the proper wording should be:
C:\Python37-32\Scripts>pip install thonnyapp.
The same in:
C:\Python37-32\Scripts>pip install -U thonnyapp
Your Thonny is nice!

aivarannamaa, 2021-10-30:
Thanks for the suggestion!

Actually it doesn't really matter, whether you are using local command path from right directory or absolute command path from any directory.

closed 2021-11-06 by aivarannamaa

Tovim, 2021-11-06:
I am afraid, I wasn't understood. It is not the path but the missing char ">" in your examples.

aivarannamaa, 2021-11-06:
Oh, you meant I should have included the command prompt in addition to the command?

I'm not sure this would make it more clear. I agree, that it's useful to include the command prompt if I also show the command output -- in this case it makes it clear which is input and which is output. But in this case it was only the command...