thonny/thonny #2029: Editor Bug in Thonny 3.3.10 on Raspberry Pi4

Rwolf01, 2021-11-02:
I hope this isn't a stupid question... I'm new to Pi, Thonny and Python, but it seems like the editor loses focus as a side effect of debugging.

If I am editing on line 147 in procedure fubar, 5 or 6 calls deep in my code, set a break point at the same place and start the debugger, by golly the execution pointer stops there.  Now suppose I step forward a few lines, find the problem, cuss a little and exit the debugger.

Seems like the code editor should return to either  A: where it was then the debugger was launched or B: where the execution pointer was when the debugger was stopped.  

It doesn't.  It ends up at the very first procedure call in __main__ way down at line 2369. (or whatever)  I tend to iterate a lot between editing and debugging, so it's maddening to constantly lose my code context like this.   

Am I doing it wrong?   If not, is there some deeper reason why it has to be this way?

Thanks, in advance, for making me less ignorant... :-)  

Other than that, I'm having a blast, climbing the learning curve on this stuff.

aivarannamaa, 2021-11-02:
Thank you for the report!

I have been simply overlooked this, as I don't use Thonny for serious development myself. I'll try to reproduce this and form my opinion on this.

aivarannamaa, 2022-03-24:
To be handled together with #2031