thonny/thonny #2030: Increase the usage of augmented assignment statements

elfring, 2021-11-03:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
(:point_right: Please check also for questionable change suggestions because of an evolving search pattern.)
```
[Markus_Elfring@fedora lokal]$ perl -p -i.orig -0777 -e 's/^(?<indentation>\s+)(?<target>\S+)\s*=\s*\k<target>[ \t]*(?<operator>[+\-%&|^@]|\*\*?|\/\/?|<<|>>)/$+{indentation}$+{target} $+{operator}=/gm' $(find ~/Projekte/thonny/lokal -name '*.py')
```

:crystal_ball: How will the development interests evolve further also according to update candidates in 53 lines of this software?

aivarannamaa, 2021-11-04:
Thank you for the suggestion, but I feel this is not important for me right now.

closed 2021-11-04 by aivarannamaa

elfring, 2021-11-04:
:crystal_ball: Under which circumstances would you care more for [information from the release notes for Python 2](https://docs.python.org/3/whatsnew/2.0.html#augmented-assignment "Augmented assignments")?