thonny/thonny #2034: [suggestion enh] rename a variable/function/class everywhere in one place

atesin, 2021-11-05:
hi... i know something similar could be done in menu > find and replace dialog > replace all...

... but it just replaces "text" in the current file wherever it finds (similar variable names, comments, etc) sometimes screwing the code

while we code and our code grows often we need to change reference names for a better or more intuitive one, or to avoid name confussion/collision, even throught different library/module files, and the more big gets our code the more compicated turns...

this feature is extrme useful when we need to change variable/reference names everywhere at once, i found this example

some ides like eclipse, netbeans, vscodium, etc, have a feature to change all variable/reference names everywhere (even throught different files)... this is extremely useful for easy maintaining our code while we write it and it grows, without losing the focus

i found this example on visual studio

[image]

aivarannamaa, 2021-11-05:
Thanks for the suggestion. I'll schedule this for 4.1

chandantiwari1, 2021-11-11:
I will do it

shivvamm, 2021-11-11:
i can also do it

aivarannamaa, 2023-01-26:
Will be handled under #1653

closed 2023-01-26 by aivarannamaa