thonny/thonny #2036: [suggestion] ability to customize mouse operations (right-click to paste text)

atesin, 2021-11-05:
hi... surely there are many users like me that are accustomed to call a context floating menu by right clicking the text, and instead find the right-click to paste text feature annoying sometimes....

sometimes i blindly do this to change window focus and thonny inserts random text from clipboard in random places (usually from another application that i used 20 min ago, has nothing to do with python, and dont remember what it was) ... so i have to undo or carefully find the intrusive text to manually delete it .... now i see you were inspired by putty/vim

my suggestion is to add some configurations to customize mouse operations, like add modifier keys or spawn right-click context menus (usually ctrl is for copy/paste operations and shift is for move/cut)

or tell me if these configurations can already be set by editing the configuration file directly.... can you add this "configure thonny hidden options by editing configuration file" help page? ...  thanks again

aivarannamaa, 2021-11-06:
Hmm, are you saying your Thonny pastes text simply by right-clicking? On Windows?  Actually, it's not supposed to do this on any platform.

Can you please give me the exact sequence of steps for reproducing this?

atesin, 2021-11-06:
mh.. i am not able to reproduce now.... but sometimes it happened, maybe in certain conditions
i will be observant

aivarannamaa, 2021-12-03:
Sorry for the late reply.

I wasn't aware of this middle-button paste feature. It seems to come from the UI toolkit (Tkinter).

As it works this way also in IDLE (which is also made with Tkinter), I'd leave it as is.

closed 2021-12-03 by aivarannamaa