thonny/thonny #2040: [request enh] ability to arrange tab windows

atesin, 2021-11-10:
hi.... in he case of editor window (or subwindow, area, view, pane, or whatever the official name is) tabs are arranged in the order they were opened and cannot be arranged/grouped

lets see i select a bunch of .py files and open all at once, they open in random order (i guess) and if i want to arrange hierarchical according imports or alphabetically, chronologic, whatever, i can't

this could be achieved by several ways, for example adding some options in tab context menu
[image]
or/and by adding some buttons in tab bar to move selected tab
[image]
or/and by dragging the tab with the mouse, the best (and maybe harder) option IMO, have even the chance to arrange tabs to other panes!... see for example my taskbar (courtesy of some taskbar/start menu configs and [open shell menu](https://open-shell.github.io/Open-Shell-Menu/) and [7 taskbar tweaker](https://tweaker.rammichael.com))
......
[image]

aivarannamaa, 2021-11-12:
Thank you for the suggestion!

I think I'll aim for drag-and-drop first.

aivarannamaa, 2023-11-26:
Fixed under #2737

closed 2023-11-26 by aivarannamaa