thonny/thonny #2041: Problem when deleting a file

AshDini, 2021-11-11:
It is difficult to explain so I provided some pictures:
[image]
[image]
[image]
[image]
[image]

aivarannamaa, 2021-11-12:
Thank you for the report!

Unfortunately I didn't understand how the deletion was related to the error. Also, the whole error message is not visible in the last screenshot. Please use "Copy to clipboard" button and post the error as text!

aivarannamaa, 2022-03-23:
Deleting via file dialog will be handled (or disabled) at #2044. In the meanwhile delete via file browser view (View => Files)

closed 2022-03-23 by aivarannamaa