thonny/thonny #2043: Missing Interpreter Tab in Option (Windows, 3.3.13/14)

broken-gage, 2021-11-11:
Interpreter tab does not show up in the option page. 

[image]

And the following error message shows up whenever I open Thonny. 

```
Traceback (most recent call last):
  File "D:\Thonny\lib\site-packages\thonny\workbench.py", line 514, in _start_runner
    self._runner.start()
  File "D:\Thonny\lib\site-packages\thonny\running.py", line 122, in start
    self.restart_backend(False, True)
  File "D:\Thonny\lib\site-packages\thonny\running.py", line 612, in restart_backend
    self._proxy = backend_class(clean)
  File "D:\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 86, in __init__
    self._fix_port()
  File "D:\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 99, in _fix_port
    potential = self._detect_potential_ports()
  File "D:\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 191, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "D:\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 880, in list_serial_ports
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'
```

aivarannamaa, 2021-11-12:
Thank you for the report!

Do you happen to have a file named `serial.py` in your working directory? Or, perhaps you have installed PyPI package called `serial`?

broken-gage, 2021-11-12:
> Thank you for the report!
> 
> Do you happen to have a file named `serial.py` in your working directory? Or, perhaps you have installed PyPI package called `serial`?

I don't see anything named serial.py file in the Thonny directory or anywhere. But I can see serial plug-in installed in the Thonny plug-ins page.

aivarannamaa, 2021-11-12:
> I can see serial plug-in installed in the Thonny plug-ins page.

If the name of the (distribution) package is `serial`, then this will mess up a required package called `pyserial`.

Try uninstalling this `serial` and restarting Thonny.

broken-gage, 2021-11-12:
> > I can see serial plug-in installed in the Thonny plug-ins page.
> 
> If the name of the (distribution) package is `serial`, then this will mess up a required package called `pyserial`.
> 
> Try uninstalling this `serial` and restarting Thonny.

Thanks, it works now after removing the serial package.

It is a fresh installation and I have no idea how the package got in.

aivarannamaa, 2021-11-13:
Most likely the package came from user site packages directory, which is under your home directory and shared among different Python interpreters of the same version.

closed 2021-11-13 by aivarannamaa