thonny/thonny #2045: systemwide linux installation

helge42, 2021-11-12:
I tried 
`sudo bash thonny-3.3.13.bash `
as mentioned [here](https://github.com/thonny/thonny/wiki/DeploymentOptions)

But this installs to /root/apps/...

I want a systemwide installation for all users. 
I use Debian 11 (Gnome) and the package 3.2.7-1  there has the autocomplete bug.

aivarannamaa, 2021-11-13:
You need to download and extract https://github.com/thonny/thonny/releases/download/v3.3.11/thonny-3.3.11-x86_64.tar.gz (assuming 64-bit machine) and pass your desired target directory to setup.py, eg

```bash
./setup /opt
```

The installer should create shared .desktop files if the target directory is outside of current user's home

helge42, 2021-11-18:
Thank you for the hints. So I could install for all users.

The debian package from sid works in bullseye, too. Just download and install it with dpkg -i .

aivarannamaa, 2021-11-19:
> The debian package from sid works in bullseye, too. Just download and install it with dpkg -i .

Thanks, that's good to know!

closed 2021-11-19 by aivarannamaa