thonny/thonny #2052: debugger problem with match and Python 3.10

no-one, 2021-11-19:
Hello,

this code is valid with Python 3.10 but it will crash the debugger:

```python
value = "foo"

match value:
    case "baz":
        print("baz")
    case _:
        print("...")
```
error message:
```
ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Traceback (most recent call last):
  File "C:\Users\test\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1279, in execute_source
    statements = compile(root, filename, mode)
ValueError: patterns may only match literals and attribute lookups
```

aivarannamaa, 2021-11-19:
Thank you for the report!

Does the same program _run_ properly in Thonny?

no-one, 2021-11-20:
Yes

aivarannamaa, 2021-12-04:
Fix will be in the first beta of Thonny 4.0

closed 2021-12-04 by aivarannamaa