thonny/thonny #2053: Debugger

vanusha1991, 2021-11-20:
I ask for help.
I noticed a pattern, when working with **indexes**, an error constantly pops up in the debugger.
The code itself works without issue.
Translation with google.
**code**:
s = input()
a = s.find('h')
b = s.rfind('h')
s = s[:a] + s[b + 1:]
print(s)

**The first error appears at this stage.**
[image]

**error text**
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1712, in event_generate
    handler(event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1213, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 278, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 407, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 590, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 466, in _update_this_frame
    self._expression_box.update_expression(msg, frame_info)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 671, in update_expression
    and msg["exception_info"]["is_fresh"]
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 799, in _highlight_range
    self.text.tag_add(tag, start_index, end_index)
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 3346, in tag_add
    (self._w, 'tag', 'add', tagName, index1) + args)
_tkinter.TclError: bad text index "4_6"

aivarannamaa, 2021-11-25:
Thank you for the report!

Which Thonny version are you using? I could not reproduce this in 3.3.14

vanusha1991, 2021-11-25:
[image]

aivarannamaa, 2021-11-25:
Really strange... Does it happen every time you try to step through this program?

vanusha1991, 2021-11-25:
Right now in this code on line # 4 this error appears. The error comes out 6 times!
numbers = [2, 6, 3, 14, 10]
print(len(numbers))
print(numbers[-1])
print(numbers[::-1])
print('YES' if 5 or 17 in numbers else 'NO')
print(numbers[1:-1])

[image]

aivarannamaa, 2021-12-25:
I now tried in Windows and still can't reproduce. Can you reproduce this if you step through the code with small steps (F7) only?

vanusha1991, 2021-12-25:
This error appears only when working with slices.
It doesn't matter how long and complex the code is. Wherever there is a slice [: n], the 'DebuggerResponse' window appears.
Since I am just learning to program, it is important for me to understand what happens during code execution, so I only use step F7.
I can give access to my computer through the Teamviewer application or record a video from the desktop.
What shall we do?
[image]

aivarannamaa, 2022-03-23:
Please see, whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better for you

JoshuaLawrence-1, 2022-08-19:
Hi there!
I also had this issue whenever using slicing, but I found a way to fix it!
In the line which you use the slicing [: n], just add a 0 before the colon like: [0:n].
This seems to work!