thonny/thonny #2055: Documentation string of functions is not shown in shell

speedy-10, 2021-11-21:
Hi, I'm not shure if this relates to issue #121.
In the traditional Python IDLE I am used to see the documentation, if I call the defined function in the shell, see picture. Now in thonny I'm missing this feature.
It would be nice to have this again in future with thonny.

Regards
[image]

aivarannamaa, 2021-11-21:
Thank you for the suggestion! This feature will appear in Thonny 4.0

closed 2021-11-21 by aivarannamaa

speedy-10, 2022-03-30:
I'm afraid the documentation string isn't implemented as I suggested:
First I activate Tools > Options > Editor "Automatically show parameter after typing '(' "
1) If I type a built-in function, e.g. "int(" the parameters are shown, that's not what I wanted (I find it annoying) -> thonny_4.0.0b1-built-in.png
2) if I type a self defined function, e.g. "ggt(", the parameters (x,y) are shown but not the intended documentation string like in IDLE see picture above 
3) the documentation line would be nicer, if it would be presented under the actual line in the shell (like in IDLE-editor)

This was tested with thonny 4.0.0b1 with ubuntu 20.04.

If you'll have time for such minor things :-) in future

Regards

[image]
[image]

aivarannamaa, 2022-04-01:
Thanks for the input!

> If I type a built-in function, e.g. "int(" the parameters are shown, that's not what I wanted (I find it annoying)

I guess I'll leave this as is, because it's not clear where to draw the line of providing or not providing the calltips.

> if I type a self defined function, e.g. "ggt(", the parameters (x,y) are shown but not the intended documentation string like in IDLE see picture above

I did not intend to show the documentation together with the calltips. Most IDE-s only show parameter info in this box. Documentation is currently available via the completions box.

> the documentation line would be nicer, if it would be presented under the actual line in the shell (like in IDLE-editor)

Agreed, the placement of the box isn't ideal yet. I'll deal with this in a later beta.