thonny/thonny #2057: Internal Error - Macos - connecting to RPI of SSH - AttributeError: 'NoneType' object has no attribute 'send_program_input'

WGandy, 2021-11-22:
I'm seeing a window pop up with a traceback after I type in the password for the SSH connection OR if I try to open a file from the RPI in Thonny.  If I close the pop up window, it does appear to connect to the RPI. 

And then after a couple restarts of Thonny, the problem went away! 

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/__init__.py", line 198, in launch
    bench.mainloop()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1283, in mainloop
    self.tk.mainloop(n)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/tktextext.py", line 46, in _dispatch_tk_operation
    return f(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 907, in intercept_insert
    self._try_submit_input()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 1089, in _try_submit_input
    self._submit_input(submittable_text)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 1231, in _submit_input
    get_runner().send_program_input(text_to_be_submitted)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/running.py", line 319, in send_program_input
    self._proxy.send_program_input(data)
AttributeError: 'NoneType' object has no attribute 'send_program_input'
```

aivarannamaa, 2021-11-22:
Thank you for the report!

Were you, by chance, typing something at the Shell while you got this error?

WGandy, 2021-11-23:
No. I was not typing anything in the Shell. A couple times I got this right after starting Thonny - right after I typed the SSH password to connect to the RPI in that pop up that popped up as it opened.