thonny/thonny #2062: error cpython.app

DocObvious, 2021-11-27:
Hi

I am trying to run some code that works without issue on my pi 4 but when I try on a new PiZeroW2 with fresh sd card I get an error:

aivarannamaa, 2021-11-28:
Hi!

It looks like you have tried to post an image but it is not visible. Better copy paste the error text.

DocObvious, 2021-11-28:
Apologies have updated the original post with the error. Thanks

aivarannamaa, 2021-11-30:
Does the error message consist of this single line you sent me? No stack-trace?

DocObvious, 2021-11-30:
I don't recall a full trace back but I will double check. Thanks

DocObvious, 2021-12-26:
Really sorry I haven't updated this the full error is as follows:

 * Serving Flask app "app" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * Restarting with stat
/usr/bin/python3: No module named thonny.plugins.cpython.app

yichenemma, 2022-01-07:
maybe try to run the program via terminal instead of Thonny?

commit a6bafccf (2022-03-24): Fix main module metadata, fixes #2062, #1782, #1769
closed 2022-03-24 by aivarannamaa