thonny/thonny #2063: Does not work in beta Armenian and shows errors

narektigranyan, 2021-11-29:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 514, in _start_runner
    self._runner.start()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 122, in start
    self.restart_backend(False, True)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 612, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 284, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 40, in __init__
    super().__init__(clean, executable)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 877, in __init__
    self._start_background_process(clean=clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 944, in _start_background_process
    **extra_params
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

narektigranyan, 2021-11-29:
I have tried to run with administrative privileges, but the same errors

aivarannamaa, 2021-11-30:
Thank you for the report!

Please open the configuration file (Tools => Open Thonny data folder => configuration.ini) and look for `backend_name` in `run` section and, in case this is `CustomCPython`, also `path` in `CustomInterpreter` section. If you have selected custom interpreter, does the path point to existing executable?

aivarannamaa, 2021-11-30:
If you don't see anything suspicious, then please close Thonny and rename configuration.ini to something else and start Thonny again. If it works this time, then there is problem with configuration and I'd be grateful if you sent me the bad configuration file.

aivarannamaa, 2022-03-23:
Please reopen if you still have this problem

closed 2022-03-23 by aivarannamaa