thonny/thonny #2064: How to clear SSH keys??

WGandy, 2021-11-29:
I've been swapping RPI units and SD cards around and it's been working ok. Until a moment ago I assumed that Thonny was using the system known hosts file (macos) but I just ran into a problem where Thonny is confused about which key is at which ip address and won't connect to the RPI I'm trying to access. This suggests to me that Thonny is keeping it's SSH keys somewhere other than .ssh/known_hosts since the keys in that file are valid. 

Is there a way to clear the keys in Thonny and start over?

aivarannamaa, 2021-11-30:
Thank you for the report!

I assumed that Thonny's SSH library does use system default configuration files (https://docs.paramiko.org/en/stable/api/hostkeys.html#paramiko.hostkeys.HostKeys.load).

Could you add more details about Thonny's confusion?

WGandy, 2021-11-30:
I was connected to a raspberry pi device via SSH from a Macos terminal window. The IP was xxx.xxx.xxx.183.  When I tried to connect to the same device from Thonny it gave me an error saying that Thonny expected a particular key but the device had returned a different key. I could see that the key that Thonny expected was associated with a different IP address in my known_hosts file. 
(Another pi device had been used previously at IP xxx.xxx.xxx.194 so I had entries for both IP addresses in my known_hosts file.)
When I swap a micro SD card to the opposite device, the card keeps the same key. I don't always keep track of which one is which but, with MacOS terminal, simply erase the key from the known hosts file and approve the new key for that IP when connecting via SSH and the OS adds that key to the known hosts file. Until this incident it seemed like Thonny was keeping up with this.  
But in this instance it was clear that what MacOS was doing was somehow different from what Thonny was doing. I'm afraid that I don't quite know how to recreate this and unfortunately, I don't have much time to spend on it either. 

What triggers Thonny to re-read the known hosts file?  Restarting didn't appear to "refresh" it.

aivarannamaa, 2021-12-03:
Unfortunately I don't know, I haven't gone deeply enough into Paramiko. 

I suggest you grep through your home directory to find out, where this information gets stored.

WGandy, 2021-12-07:
Ok, I think I may have figured this out. 
MacOS is willing to look at all of the entries in the known_hosts file for a given IP address and allow any of the keys that are listed. Thonny appears to only look at the first entry in the known_hosts file for the IP address in question. I found that commenting out earlier entries (with the same IP) such that the only entry listed is the one with the key I am currently trying to use, will make Thonny happy enough to connect.

aivarannamaa, 2021-12-09:
Thank you for this information!

It looks like the library Thonny is using doesn't allow customizing this behavior (https://docs.paramiko.org/en/stable/api/hostkeys.html) so I guess it needs to stay like this.

closed 2021-12-09 by aivarannamaa

hapeldoorn, 2022-01-05:
Hello, I have exact the same problem but on a windows enviroment. I succeeded to secure putty acces with ssh keys & passphrase. Logging in that way via putty is no problem. But using Thonny deny's using that. Reverting to normal operations is simply not possible. I removed the SSH key option with succes in Putty and removed the content of .ssh directory at the pi. Thonny is still expecting SSH keys even when I tell it to simply use password challenge auth. A reinstall of Thonny didnt't solve the issue.  Searching for  known_hosts on windows and pi didn't gave 1 result / hit. Also searched the windows register for that.

I can realy need some advise to proceed, currently I can't use the pi for python programming with thonny. Hope one can help me out