thonny/thonny #2071: ESP32: How to hard reset ESP32 via the RTS pin of USB/COM converter CP2102?

IhorNehrutsa, 2021-12-08:
Sometimes I make mistakes in my code wich break REPL communication. 
Is Thonny has command for hardware reset of chip, like the **esptool**?

https://github.com/espressif/esptool/blob/master/esptool.py#L585-L588
```
            self._setRTS(True)   # EN=LOW, chip in reset
            time.sleep(0.1)
            self._setDTR(True)   # IO0=LOW
            self._setRTS(False)  # EN=HIGH, chip out of reset
```
Thank your.

aivarannamaa, 2021-12-09:
Thank you for the request!

No, Thonny does not have such feature. Do you specifically want a command for hard reset or the ability to automatically reconnect after hard reset?

IhorNehrutsa, 2021-12-09:
Yes I want a command(to make a keyboard shortcut) for hard reset the microcontroller and then automatically reconnect Thonny after hard reset.

Thank your very much.

bnelsey, 2022-03-29:
for hard resetting the microcontroller, sending these two commands does the job:
```
import machine
machine.reset()
```
if we can get a shortcut to send those two commands so we don't have to type them out each time would be awesome

IhorNehrutsa, 2022-03-30:
@bnelsey 
machine.reset() is not a hard reset, only a soft one. 
It will work when REPL can get and perform the command.
It does not work when the software is locked in an infinite loop or a blocking operation is in progress.

aivarannamaa, 2024-05-30:
The problem is that different ESP-s on different platforms have different ways of invoking the hard reset. I'm afraid I don't have time for this feature.

closed 2024-05-30 by aivarannamaa

IhorNehrutsa, 2024-05-30:
I suggest moving this task to the 5.1... or 6.0... milestone.
Thanks.