thonny/thonny #2072: [BUG] Filedialog false background color

ckvsoft, 2021-12-09:
When using the dark themes the Open Filedialogs background color has same color as the textcolor.
[image]

lg Chris

aivarannamaa, 2021-12-09:
Thank you for the report! Which Linux distro and which desktop environment are you using?

ckvsoft, 2021-12-10:
Oh sorry.
Kubuntu 21.10 KDE

aivarannamaa, 2021-12-25:
I now installed Kubuntu 21.10 to a VM but I can't reproduce it.

You're using dark system theme and Clean Dark Thonny theme, right?

ckvsoft, 2021-12-26:
Breeze dark, and in Thonny I also tried raspberry dark. It doesn't matter what settings I try. In the dark mode, the file open dialog always has a white background and white text. Got this on my laptop and also on a desktop with Kubuntu 20.04

But I know that. I program a cash register in c ++. The users often have errors that I can hardly understand here. But don't worry about it. It works that way too ;-)

aivarannamaa, 2022-01-11:
Didn't just re-starting Thonny achieve the same effect?

ckvsoft, 2022-01-11:
Of course not. Do you think I haven't done that since the Thonny installation?

aivarannamaa, 2024-06-15:
Will be fixed under #2756

closed 2024-06-15 by aivarannamaa