thonny/thonny #2077: [feature suggestion] option to save file as .py when no extension provided

LovelyA72, 2021-12-11:
It will be nice to have such a feature exists since it's impossible to rename a file in CircuitPython device right now and accidentally creating a file with no extension can be a bit frustrating to deal with.

aivarannamaa, 2021-12-12:
Thank you for the suggestion!

The "Rename" command is indeed missing in the file browser (#1429), but you should be able to rename currently open file via "File => Move/Rename".

Instead of automatically appending the `.py` extension, I plan to add a confirmation dialog (#1874). This way there would be less magic and no need for a new option.

aivarannamaa, 2022-01-31:
In the upcoming Thonny 4.0, when you omit an extension, you'll get following dialog:

[image]

closed 2022-01-31 by aivarannamaa