thonny/thonny #2078: Thonny

mikedabb6, 2021-12-12:
I have been using Thonny for quite a while now on a Raspberry Pi 400 but find Thonny a bit flaky to use.

It does not always disconnect or connect to the Pico correctly and sometimes just crashes with no error messages.

I spend a lot of time just unplugging and plugging the Pico back in to get Thonny to work with it.

Even just changing a small amount of code stops the Pico from working correctly.

aivarannamaa, 2021-12-13:
Thank you for the report! I need more specific information for improving things, though.

The MicroPython support will be refactored in Thonny 4.0, so this may fix your issues. In the meanwhile let me know if you find repeatable steps for reproducing your issues.

mikedabb6, 2021-12-13:
Hi,  thanks for the quick reply.

I have some code that repeats indefinitely and changes the colour of circles on the screen. I changed the RGB colour from 255 to 0 and the code only looped for about 6 times and stopped working. I changed it back and it worked again. This morning I tried it again and the same happens but this time I un-commented out a debug print statement to see how many times the loop ran for and it worked fine with RGB number change. It is vey strange behaviour that I cannot decide why this should happen when all of the syntax is correct.

Regards
Mike

aivarannamaa, 2021-12-15:
> I un-commented out a debug print statement to see how many times the loop ran for and it worked fine with RGB number change.

Can you post the code as it was when it caused problems?

mikedabb6, 2021-12-16:
Hi Aivar

I commented out the debug print statement at the bottom of the code 'print(pattern_counter)' and it

would not run as before. I then put the debug print statement back in and for

some reason it would not run now. It really seems to be random what will cause the code to fail.

Please see the code below:

import picodisplay2 as display
import utime, random

width = display.get_width()
height = display.get_height()

display_buffer = bytearray(width * height * 2) # 2-bytes per pixel (RGB565)
display.init(display_buffer)
display.set_backlight(1.0)
display.set_pen(0, 0, 0)
display.clear()

display.set_pen(0, 0, 0)
display.update()
radius = 10
num_circles = 154
#num_squares = 96
pen_colour = 0,0,0
pattern_counter = 0
digit_1 = 0
digit_2 = 0
digit_3 = 0

class FlashCircle:
def __init__(self, centre_x, centre_y, radius, pen_colour):
self.centre_x = centre_x
self.centre_y = centre_y
self.radius = radius
self.pen_colour = pen_colour

while True:
flashcircle = []
centre_x = 12
centre_y = 10
counter = 0
my_colourlist = ["A","B"]

for i in range(0, num_circles):
choice = random.choice(my_colourlist)

if (pattern_counter >= 0 and pattern_counter <= 30):
if choice == 'A':#red
digit_1 = 255
digit_2 = 0
digit_3 = 0
if choice == 'B':#white
digit_1 = 255
digit_2 = 255
digit_3 = 255

if (pattern_counter > 30 and pattern_counter <= 60):
if choice == 'A':#yellow
digit_1 = 255
digit_2 = 255
digit_3 = 0
if choice == 'B':#blue
digit_1 = 0
digit_2 = 0
digit_3 = 255

if (pattern_counter > 60 and pattern_counter <= 90):
if choice == 'A':#white
digit_1 = 255
digit_2 = 0
digit_3 = 0
if choice == 'B':#green
digit_1 = 0
digit_2 = 255
digit_3 = 0

flashcircle.append(
FlashCircle(
centre_x,
centre_y,
radius,
display.create_pen(digit_1, digit_2, digit_3),
)
)
counter += 1
centre_x += 22
rand = random.randint(0, 1)

if counter == 14:
centre_x = 12
centre_y = 32
if counter == 28:
centre_x = 12
centre_y = 54
if counter == 42:
centre_x = 12
centre_y = 76
if counter == 56:
centre_x = 12
centre_y = 98
if counter == 70:
centre_x = 12
centre_y = 120
if counter == 84:
centre_x = 12
centre_y = 142
if counter == 98:
centre_x = 12
centre_y = 164
if counter == 112:
centre_x = 12
centre_y = 186
if counter == 126:
centre_x = 12
centre_y = 208
if counter == 140:
centre_x = 12
centre_y = 230
if counter == 154:
centre_x = 12
centre_y = 252

for i in range(0, num_circles):
flashcircle_obj = flashcircle[i]
display.set_pen(int(flashcircle_obj.pen_colour))
display.circle(flashcircle_obj.centre_x, flashcircle_obj.centre_y,
flashcircle_obj.radius)

display.update()
utime.sleep(0.3)
print(pattern_counter)
if pattern_counter == 90:
pattern_counter = 0
pattern_counter += 1

aivarannamaa, 2022-03-23:
Please see whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works better for you

closed 2023-01-28 by aivarannamaa