thonny/thonny #2080: Linux install not clean in user mode

mikamyara, 2021-12-13:
Dear all,

  I work with Thonny on Linux usually on a Mac with a Raspberry Pico and it works like a charm. I want to use it with students in a room with Linux (Ubuntu) Mint installed. I installed it on Linux using the ubuntu package (sudo apt install thonny) and the Raspberry Pico was not recognized. Thonny also displayed python errors (from its own code) when opening package manager. However Thonny worked with local (host) Python.

  I uninstalled the package and downloaded from thonny.org : same problem, with a standard user. Then I tried with root and everything works with the root user. But more : if I move Thonny installed in /root/apps/thonny to for example /opt/thonny , then I update the /usr/share/applications/ icon for Thonny, then everything works for any user. In the meantime I added rules in the udev but I am not sure it was useful.

  I guess it is a problem for most linux users, and it is easy to solve (installation problem only). I am not sure it is specifically related to Raspberry Pico but I have no mean to explore other configurations

  Thanks a lot,
    Mikhaël

aivarannamaa, 2021-12-15:
Thank you for the report!

> I installed it on Linux using the ubuntu package (sudo apt install thonny)

Ubuntu's package repo has too old Thonny version.

> Thonny also displayed python errors (from its own code) when opening package manager.

It would be helpful if you could post the error together with stacktrace.

At the moment I don't have any idea what could have been the problem and how to fix this.

aivarannamaa, 2022-03-23:
You could try https://github.com/thonny/thonny/releases/tag/v4.0.0b1

aivarannamaa, 2023-01-29:
Please reopen if you still have this problem with Thonny 4.0.2

closed 2023-01-29 by aivarannamaa